kruzhok-team/lapki-client #61: Порядок отрисовки нод

bryzZz, 2023-10-19:
Сейчас ноды отрисовываются в порядке перечисления. Нужно сделать так, чтобы этот порядок можно было менять.
При нажатии на состояние или связь делать его поверх остальных.
[image]

bryzZz, 2023-10-20:
Вариант схемы со слоями:
[image]

Этот вариант рабочий но у него есть одна проблема. Допустим у нас есть две ноды, при выделении первой она переместилась на второй слой и стала поверх второй ноды. Что делать при выделении второй ноды? Перемещать на третий слой? Или перемещать первую ноду обратно на первый слой а вторую на второй?

Тут хотелось бы какое-нибудь поведение массива, чтобы выделенная нода перемещалась в конец и рисовалась последней(то есть самой верхней). Этот вариант сделать не получается потому что в системы есть очень много операций чтения и не хотелось бы терять сложность O(1). Ещё в таком варианте не понятно как сохранить эту последовательность при перезагрузке, в первом варианте в схему можно просто добавить слои.

Есть идея ввести кэш порядка отрисовки, чтобы он был массовом айдишников и реагировал на изменения объекта с нодами чтобы изменится самому. Тогда во время отрисовки нужно будет итерироваться по кэшу и выдёргивать ноды по айдишнику. Ещё лучше чтобы кэш был связным списком, тогда при перемещении ноды в конец не нужно создавать новый массов

chekoopa, 2023-10-23:
По-хорошему, «слои» есть как минимум в пределах дочерних контейнеров. Кажется, наиболее понятной историей здесь будет порядок «вложенность – z-order – идентификатор». То есть дочерние ноды имеют базовую вложенность, как у своей родительской ноды, в её пределах смотрим на z-order (условно, номер слоя), при одинаковом z-order смотрим на id-шник. 

Другой вопрос, как это с инженерной стороны реализовать. Возможно, стоит пошариться по похожим решениям и посмотреть, как сделано у них.

PR #62 (2023-10-24): Очередь рендера
Closes #61

Тут просто добавлен DrawList, на основе которого происходит рендер
PR #68 (2023-10-26): Render queue refactor
Closes #61 

эээээээээээм, нууууууууууу, странная штука, я бы даже назвал это говнокодом. Что изменилось:
- Draggable стал Node
- Transitions стал TransitionsController
- States стал StatesController
- StateMachine стал MachineController
- Клики теперь отслеживает только Container и передаёт вниз, а не вверх из Draggable как раньше
- Transition стал вкладываться в состояния
- Condition удалился, теперь всё в Transition для удобства поставки его в очередь отрисовки
- У переходов больший приоритет на рендер (они всегда выше состояний)
- У Container и Node есть поле children [... cut]
closed 2023-10-24 by chekoopa