kruzhok-team/lapki-client #149: Редизайн модальных окон

XidFanSan, 2023-12-11:
Требуется переписать как минимум createModal.tsx для лучшей читабельности и разделения кода для разных задач, потому что сейчас всё реализовано в одном файле.

chekoopa, 2023-12-11:
Пожалуйста, добавь предположение о том, как именно нужно переписать CreateModal. 

Да, сейчас модалка одновременно отвечает за ноду и переход, но у них по факту много общего в логике (выбор триггера и набора действий, а в перспективе условия будут распространяться и на внутренние события). Разделение модалки на две разных может и облегчит логику по данным, но у нас появится ещё одна модалка, на 70% дублирующая другую. Так себе дополнение к и без того загруженному интерфейсу, как мне кажется. 

@bryzZz Твоё мнение?

bryzZz, 2023-12-12:
По моему нужно сделать отдельные компоненты общих блоков(например блок добавления действий) и вставить их в разные модалки

XidFanSan, 2023-12-12:
Поддерживаю Никиту

chekoopa, 2023-12-12:
Я не против вынести компоненты в отдельные классы, но использование отдельных модалок меня беспокоит, это будет уже седьмое модальное окно. С другой стороны, если оно будет также прятаться за тернарным оператором... 

Попробуйте тогда и проверьте по производительности.

chekoopa, 2024-06-04:
Процесс пошёл, вынесли компоненты в хуки. Пока выглядит адекватно.

chekoopa, 2024-06-04:
Компоненты вынесены в хуки, процесс пошёл. Дальше только дорабатывать и совершенствовать.

closed 2024-06-04 by chekoopa