kruzhok-team/lapki-client #261: Рефактор классов Compiler и Flasher

Roundabout1, 2024-04-12:
Compiler и Flasher имеют схожую функциональность:
1. Подключение по веб-сокету к серверу по указанному адресу
2. Смена сервера
3. Прямое переподключение в случае потери соединения с сервером
4. Автоматические попытки переподключения к серверу в случае потери соединения

Несмотря на наличие одинаковых методов, каждый из классов реализует своё решение, было бы намного удобнее и надёжнее работать с этими классами, если бы они наследовали одинаковую функциональность через родительский класс.

chekoopa, 2024-04-19:
Ну да, настройка и поддержание подключения к серверу – это достаточно общая функциональность. Возможно, есть элегантное решение (может, даже на callback-ах)

chekoopa, 2024-08-13:
Закрыто в #361.