kruzhok-team/lapki-client #315: Пробелы в названиях компонентов

captaincod, 2024-06-02:
IDE никак не указывает, что нельзя создавать компонент с названием, в котором содержится пробел. Если такой создать, то либо IDE через неопределённое время крашится с ошибкой undefined, либо схема после сохранения и повторного открытия выдаёт, например, ошибки, как на скриншоте (см. переходы)

[image]

Прикрепляю ссылку на схему, с которой срабатывает ошибка:
https://gist.github.com/captaincod/b46a683bb67dd82bbcf839e557e19956

chekoopa, 2024-06-03:
Нужно либо запрещать пробелы (что логично для компонентной архитектуры, особенно в Arduino), либо как-то их обрабатывать и экранировать. Склоняюсь к первому варианту (можно в мягкой форме, заменять на подчёркивания)

Roundabout1, 2024-07-09:
Можно закрыть issue

closed 2024-07-10 by chekoopa