kruzhok-team/lapki-client #382: Ошибка при отмене импорта

captaincod, 2024-08-07:
При нажатии кнопки "Импорт" в главном меню и последующей отмене импорта (не выбрать ни один файл, нажать "Отмена") появляется экран с ошибкой и следующее сообщение в консоли:
```JS
TypeError: Cannot read properties of null (reading 'getComponent')
    at getComponentOption (index-e1dfc5b0.js:67612:41)
    at index-e1dfc5b0.js:67622:22
    at Object.useMemo (index-e1dfc5b0.js:4143:8)
    at react_production_min.useMemo (index-e1dfc5b0.js:321:22)
    at useTrigger (index-e1dfc5b0.js:67610:41)
    at StateModal (index-e1dfc5b0.js:68115:19)
    at Xh (index-e1dfc5b0.js:3791:8)
    at Wk (index-e1dfc5b0.js:6887:12)
    at Vk (index-e1dfc5b0.js:6554:12)
    at Uk (index-e1dfc5b0.js:6547:5)
```

[image]

chekoopa, 2024-08-07:
Вот здесь обработка неправильно проходит, надо изучить.

https://github.com/kruzhok-team/lapki-client/blob/c8795ab767654e7dddb71bb5aa772ee310998fea/src/renderer/src/lib/data/EditorModel/FilesManager.ts#L80-L86

L140-beep, 2024-08-12:
Почему-то platform в контроллере равно null в момент, когда мы пытаемся к нему обратиться, добавление доп условия не решает проблему

L140-beep, 2024-08-13:
Пофикшено в #395

closed 2024-08-13 by chekoopa