kruzhok-team/lapki-client #457: «Нормализация» геометрии схем

chekoopa, 2024-10-02:
Необходимо найти общую точку по геометрии со стандартом НКФП.ПРИМС и принятым там визуальным представлением схем.

Как минимум, это значит, что дочерние состояния должны перекрывать контейнер с событиями.

L140-beep, 2024-10-10:
@chekoopa, За перекрытие отвечает вот эта функция в Shape.ts. Я предлагаю добавить абстрактный getter border, переопределяемый каждым Drawable.
```
  handleMouseMove = (e: MyMouseEvent) => {
    // код...
    if (this.parent) {
      this.position = {
        x: Math.max(0, this.position.x),
        y: Math.max(0, this.position.y),
      };
    }
  };
```

```
get border() {
  return { x: 0, y: 0 };
}

  handleMouseMove = (e: MyMouseEvent) => {
    // код...
    if (this.parent) {
      this.position = {
        x: Math.max(this.parent.border.x, this.position.x),
        y: Math.max(this.parent.border.y, this.position.y),
      };
    }
  };

```

chekoopa, 2024-10-11:
Речь даже не про border, а про то, что по факту контейнер с событиями должен быть частью дочернего контейнера, и координаты должны считаться от левого верхнего угла (либо под заголовком, либо над, если ориентироваться на стандарт ПРИМС).