kruzhok-team/lapki-client #460: Сборка для Windows 11. Пустое окно приложения

DmitryPanov78, 2024-11-01:
Sub items Errors: 2

                    errorOut=ERROR: Cannot create symbolic link : ������ �� �������� �ॡ㥬묨 �ࠢ���. : C:\Users\reacher\AppData\Local\electron-builder\Cache\winCodeSign\398400455\darwin\10.12\lib\libcrypto.dylib
    ERROR: Cannot create symbolic link : ������ �� �������� �ॡ㥬묨 �ࠢ���. : C:\Users\reacher\AppData\Local\electron-builder\Cache\winCodeSign\398400455\darwin\10.12\lib\libssl.dylib

                    command='c:\Users\reacher\lapki-client-main\lapki-client-main\node_modules\7zip-bin\win\x64\7za.exe' x -bd 'C:\Users\reacher\AppData\Local\electron-builder\Cache\winCodeSign\398400455.7z' '-oC:\Users\reacher\AppData\Local\electron-builder\Cache\winCodeSign\398400455'
                    workingDir=C:\Users\reacher\AppData\Local\electron-builder\Cache\winCodeSign
  • Above command failed, retrying 0 more times
[image]

chekoopa, 2024-11-03:
Доброго дня. Прошу уточнить:
- версию системы
- версию Lapki IDE
- порядок действий, который привёл к ошибке.

chekoopa, 2025-02-02:
Если ошибка проявится вновь, просьба создать новое сообщение об ошибке, включив информацию выше.

closed 2025-02-02 by chekoopa