kruzhok-team/lapki-client #506: Очистка действий в состоянии при выполнении условия

Roundabout1, 2024-11-26:
В ситуации, как на скриншоте нельзя удалить действие через редактор состояния, если оно является единственным. Однако, это можно сделать через контекстное меню на канвасе.
[image]

L140-beep, 2024-11-28:
@chekoopa Я бы сказал, что проблема тут в другом - мы сознательно запрещаем пользователю оставлять события пустыми. Это еще с самого начала IDE такое поведение. Стоит уточнить, что удалить последнее действие можно, но сохранить, не добавив новое, не получится. Если же мы удаляем последнее действие на канвасе, то вместе с ним удаляется и событие. Предлагаю поменять название issue на `"Добавить события без действий"` с пометкой `enhancement`

chekoopa, 2025-02-09:
@L140-beep @Roundabout1 Кажется, больше не актуально с новым редактором событий.

Roundabout1, 2025-02-10:
Да, новый редактор решает эту проблему

closed 2025-02-10 by Roundabout1