kruzhok-team/lapki-client #618: Неинициализированный пин ReDe вешает пользовательскую шину

regellosigkeitsaxiom, 2024-12-23:
Если этот пин не инициализировать и бросить, то трансивер пользовательской шины включается в режим передачи (при условии, что киберген передачу разрешает, а он по-умолчанию разрешает) и вешает всю пользовательскую шину.
Чтобы этого не было, этот пин должен быть инициализирован всегда, даже если конкретный модуль пользовательскую шину не использует.
Параметры инициализации: Open-drain (OD), прижать к земле.

CodeR-na-r1, 2024-12-25:
В прошивке пользователя пин уже инициализируется (https://github.com/kruzhok-team/lapki-compiler/commit/19de6df6783795d6eff56f9ed2a27f4bad33f5f1)
Есть возможность инициализировать пин из загрузчика, сказав кибергену?

closed 2024-12-25 by chekoopa

regellosigkeitsaxiom, 2024-12-25:
Нет, такой возможности нет, Будем думать над аппаратным решением в будущих платах.