kruzhok-team/lapki-client #629: Названия компонентов не допускают кириллицы

regellosigkeitsaxiom, 2025-01-11:
[image]

Поддержка кириллицы должна быть полной.
Также не очень понятно, почему в этом названии нельзя использовать пробелы.

L140-beep, 2025-01-13:
@regellosigkeitsaxiom Потому что название компонента вставляется в код "как есть".

chekoopa, 2025-01-14:
Пробелы – точно не можем, потому что чтение имени компонента станет невозможным.

Про кириллицу – она была раньше. Предложения, как это вернуть:
1. добавить культурную подсказку (почему именно такое ограничение)
2. допустить кириллицу в названии (если это позволяет целевая платформа)
3. разделить человеческое и техническое имена. Техническое делать через параметр, можно опциональный (см. ниже)
4. сделать имя человеческим генерировать техническое имя на его основе (транслитерация + хэш на 4-6 символов в конце)

Предлагаю сейчас сделать пункт 1 и двигаться в сторону 4.

closed 2025-02-06 by chekoopa