kruzhok-team/lapki-client #923: Импорт и экспорт из Защиты пасеки

Roundabout1, 2025-05-13:
Игра "Берлога: Защита пасеки" для Windows (на других платформах не проверялось) сохраняет файлы со схемами в расширении `.xml` вместо `.graphml`. При этом IDE может импортировать и экспортировать только `.graphml` файлы.

Если поменять расширение файла с .xml на .graphml, то импорт все равно не заработает, метод выдаст ошибку, которая видна только в консоле:
```
caught TypeError: Cannot read properties of null (reading 'initialState')
    at actualizeElements (index-73acbcc0.js:43964:75)
    at Compiler.messageHandler (index-73acbcc0.js:44483:20)
    at ws2.onmessage (index-73acbcc0.js:44291:12)
```

chekoopa, 2025-05-15:
@L140-beep

L140-beep, 2025-05-22:
В обновлении поменяли конвенцию наименования файлов (на `имя_тип.xml`) и алгоритм импорта схем в Берлогу. Если первый пункт уже дело решенное, то по поводу экспорта нужно запрашивать информацию у разработчиков Берлоги